sadr0b0t/stepper_h #3: Добавить юнит-тесты для цикла с шагами

1i7, 2016-12-08:
Отслеживать ошибки типа этой уже реально проблематично
https://github.com/1i7/stepper_h/issues/1

Нужны юнит-тесты, в первую очередь для счетчика шагов.

Главная проблема - модуль шагания завязан на API таймера PIC32 и некоторых вызовов из API Ардуино. Идея гонять тесты на контроллере мне уже не очень нравится, тем более, внедрять тестирующий код в обработчик прерываний (который и так разбух). Успевать что-то делать между тиками таймера, пожалуй, тоже не очень реально.

Короче, нужно придумать, как оптимально тестировать этот модуль. И желательно не на контроллере, а в настольной операционке.

Первая ссылка Arduino Unit testing ведет к ArduinoUnit
https://github.com/mmurdoch/arduinounit

Сразу не очень нравится, тк это набор вызовов для удобства тестирования типа ассертов для добавления в Скетч на микроконтроллере, главную проблему не решает.

Вторая ссылка на обсуждение

How can I unit test Arduino code?
http://stackoverflow.com/questions/780819/how-can-i-unit-test-arduino-code

Там отличный комент про то, что скетчи нужно тестировать на десктопе и сразу пример техники
http://stackoverflow.com/a/11437456/3466540

Предлагает ардуиновские вызовы заменять на заглушки и приводит код некоторых своих заглушек здесь
https://github.com/IronSavior/dsm2_tx

берем его за основу

1i7, 2016-12-09:
Добавить несколько заглушек к системным вызовам оказалось сильно проще, чем можно было ожидать. Всё собралось и даже запускается.

Для симуляции таймера вызываю в цикле
~~~cpp
handle_interrupts(3);
~~~

Очень удобно, т.к. после каждого "тика" таймера можно вставлять необходимые проверки и конролировать "движение" мотора на каждой итерации.

Теперь следует выбрать какой-нибудь не слишком тяжелый, а лучше вообще легкий фреймворк для юнит-тестов на С++. 

предварительные ссылки
О модульном тестировании на C++ и о CxxTest
https://habrahabr.ru/post/69160/

здесь
Boost::Test
QTestLib
UnitTest++
CxxTest

в коментах ссылки
GoogleTest 
CppUnit

еще 
http://stackoverflow.com/questions/242926/comparison-of-c-unit-test-frameworks
https://en.wikipedia.org/wiki/List_of_unit_testing_frameworks#C.2B.2B

пока остановлюсь на UnitTest++
https://github.com/unittest-cpp/unittest-cpp/wiki

https://github.com/unittest-cpp/unittest-cpp/wiki

1i7, 2016-12-09:
И еще хороший вопросник пусть тоже будет здесь, т.к. у нас весь код на С++, а инициализация таймера на Си

How to mix C and C++
https://isocpp.org/wiki/faq/mixing-c-and-cpp

если коротко,

нужно подключать сишные библиотеки как

~~~cpp
extern "C"{
    #include "../stepper_h/timer_setup.h"
}
~~~

и потом собирать сишный код и сиплюсплюсный код отдельно в объектные файлы *.o:

~~~bash
gcc -c timer_setup_stub.c -o timer_setup_stub.o
g++ -std=c++11 -I. -c WProgram.cpp ../stepper_h/stepper.cpp ../stepper_h/stepper_timer.cpp stepper_test.cpp
~~~

линковать все вместе скопом

~~~bash
g++ *.o -o stepper_test
~~~

1i7, 2016-12-10:
UnitTest++ все-таки тяжеловата - нужно компялять, устанавливать и т.п.:
https://github.com/unittest-cpp/unittest-cpp/wiki/Building-Using-CMake

Хочу вообще легкого, чтобы кинуть рядом с исходниками и все скомпилялось.

Есть еще такой вариант, выглядит полегче
cpptest-lite
https://github.com/doe300/cpptest-lite

Документации, похоже, нет, но примеры можно посмотреть в тестах самой библиотеки
https://github.com/doe300/cpptest-lite/tree/master/test

сначала не скомпилялось, но потом скомпилялось исправлением одной строки
https://github.com/doe300/cpptest-lite/issues/3#issuecomment-266192364

Под ардуино не компиляется, т.к. в 
https://github.com/doe300/cpptest-lite/blob/master/include/TestSuite.h

в заголовке подключаются библиотеки, которых нет в ардуине

~~~cpp
#include <string>
#include <vector>
#include <chrono>
#include <functional>
#include <memory> //shared_ptr
#include <type_traits> //is_constructible
~~~

1i7, 2016-12-10:
Вот это, похоже, еще легче
https://github.com/nassersala/cbdd

под ардуиной тоже не компиляется с ошибками типа
~~~
In file included from /tmp/arduino_modified_sketch_584194/sketch_dec10a.ino:3:0:
/home/benderamp/Arduino/libraries/cbdd/cbdd.h:18:41: error: expected ')' before '^' token
 void describe(const char *string, void (^block)());
                                         ^
/home/benderamp/Arduino/libraries/cbdd/cbdd.h:18:41: error: expected ')' before '^' token
/home/benderamp/Arduino/libraries/cbdd/cbdd.h:18:41: error: expected initializer before '^' token
/home/benderamp/Arduino/libraries/cbdd/cbdd.h:19:40: error: expected ')' before '^' token
 void context(const char *string, void (^block)());
~~~

плюс вывод через printf, но всего в паре мест, очень легко заменить, если сильно захочется
https://github.com/nassersala/cbdd/blob/master/src/color_terminal_displayer.c

1i7, 2016-12-10:
>под ардуиной тоже не компиляется с ошибками типа
на десктопе тоже не компилируется с этими же ошибками - это gcc, а они предлагают его собирать с  clang

1i7, 2016-12-10:
Короче, пока вот идеал:
http://www.use-strict.de/sput-unit-testing/
http://www.use-strict.de/sput-unit-testing/tutorial.html

Все умещается в одном инклуднике. В стартовом коде практически ничего лишнего. С gcc на десктопе собрался и запустился. Собрался даже для обычной Ардуины (для чипкита вылезла какая-то странная ошибка линковки, может быть, можно тоже пофиксить), только почему-то прошивка, скомпилированная с этой библиотекой, на плату Ардуино Леонардо не шьется (реально, плата отваливается и ее нужно переподключать), хотя все компиляется без ошибок и место есть. Нужно будет еще попробовать на Меге или на обычной Uno.

Вывод можно перенаправить в любой стрим: по умолчанию - в std::out, можно в файл, может на Ардуине тоже что-нибудь найдется подходящее для связи с последовательным портом.

Плюс.
При случае можно будет еще посмотреть эту либу:
https://github.com/danfis/cu

у нее всего два файла (заголовочник и сишник), но код определения и добавления тестов какой-то некрасивый.

1i7, 2016-12-19:
В общем, тесты добавили, библиотека - sput
https://github.com/1i7/stepper_h/commit/4db49e9a59ab10470714f0cc4542597ace0022b3

Пока добавлены тесты:
- жизненный цикл цикла вращения (запущен/остановлен/пауза)
- движение с максимальной скоростью (разобран каждый тик таймера)
- движение со скоростью, некратной периоду таймера (здесь сразу всплыл баг в библиотеке)
- выход за границы (баг воспроизводится)

Для старта достаточно, остальные тесты в новых тикетах.

По библиотеке sput - довольно неплохая в плане лаконичности, но даже с перебором: assertEquals в нее бы добавить не помешало (иначе не видно, какое именно неправильное значение попало в переменную - нужно добавлять дополнительные отладочные строки, чтобы его напечатать). Но зато показывает номер ошибочного теста и строку, поэтому все равно ок.

closed 2016-12-19 by 1i7

sadr0b0t, 2018-08-04:
> Первая ссылка Arduino Unit testing ведет к ArduinoUnit
> https://github.com/mmurdoch/arduinounit
> 
> Сразу не очень нравится, тк это набор вызовов для удобства тестирования типа ассертов для добавления в Скетч на микроконтроллере, главную проблему не решает.

Все-таки этот вариант тоже был вполне подходящий, у них есть режим "En Vitro" для запуска этих же тестов на десктопе (в том числе какие-то рекоменддации или заготовки для заглушек). В любом случае потяжелее, чем sput.h