sadr0b0t/stepper_h #4: Порт на классическую Arduino с AVR

1i7, 2016-12-11:
Хорошо бы сделать порт на классическую Ардуино с чипами AVR. И хорошо бы сделать этот порт так, чтобы компилировалась правильная версия бебилиотеки в зависимости от целевой платформы.

На тему поддержки разных платформ в одной библиотеке хорошее обсуждение:
resolving multiple platform libraries?
http://stm32duino.com/viewtopic.php?t=440

Основные варианты:
- вставлять в платформо-зависимый код дефайны
- укладывать библиотеки в недра ардуины типа
~~~
Arduino\hardware\ESP8266com\esp8266\libraries
~~~

В самом степпере основной платформо-зависимый код - настройки таймера здесь (сейчас работает на PIC32, такую же вирсию нужно сделать для AVR)
https://github.com/1i7/stepper_h/blob/master/stepper_h/timer_setup.c

Плюс немного настроек (выставление частоты таймера) в stepper_timer.cpp
https://github.com/1i7/stepper_h/blob/master/stepper_h/stepper_timer.cpp#L642

его лучше обернуть для универсальности, чтобы эта часть вообще не зависела от платформы.

За пример работы с таймерами на AVR можно взять ардуиновскую библиотеку Servo.h (версию для PIC32 я взял как раз из чипкитовской версии этой же библиотеки).

1i7, 2016-12-21:
В общем так, делаю разные версии timer_setup.c для каждой платфомы: 
- timer_setup_pic32.c
- timer_setup_avr.c

внутри код оборачиваю в макрос ifdef для архитектуры: если архитектура одходит, то компилять, если не подходит - не компилять.

здесь написано, что стандарта на макросы для обозначения архитектур нет

Detecting CPU architecture compile-time
http://stackoverflow.com/questions/152016/detecting-cpu-architecture-compile-time

здесь есть список разных макросов для разных архитектур в разных компиляторов, но PIC32 там нет

Pre-defined Compiler Macros
https://sourceforge.net/p/predef/wiki/Architectures/

вот здесь какое-то обсуждение, где в коди приводится макрос для PIC32MZ (и варианты для конкретных чипов)

Moved from PIC32MX to PIC32MZ, now having SPI Problems
http://www.microchip.com/forums/m964409.aspx

~~~cpp
#ifdef __PIC32MZ__
...
#endif
~~~

сделал вот так:
~~~cpp
#ifdef __PIC32__
~~~

Всё зашибись: на чипкитах код компиляется, на ардуинах с AVR вылетает с ошибкой линкера (не определены нужные функции - так и есть).

Теперь нужно добавить версию таймера для AVR

1i7, 2016-12-21:
Для PIC32 (ChipKIT) код по инциализации таймера взят из порта библиотеки Servo.h здесь
https://github.com/chipKIT32/chipKIT-core/blob/master/pic32/libraries/Servo/utility/int.c

там сейчас еще обновили код похоже, наставили дефайнов для разных чипов:

~~~cpp
#if defined(__PIC32MX1XX__) || defined(__PIC32MX2XX__)
    // 40000000 / 32 / 25000 = 50 => 20ms
    uint8_t     tckps   = 0b101;    // set prescalar 1:32
    uint16_t    prx     = 0x61A8;   // 25000
#elif defined(__PIC32MZXX__)
~~~

Можно тоже обновить код, по крайней мере, посмотреть, как запустится на WiFire с PIC32MZ.

У Ардуины эта библиотека здесь:
https://github.com/arduino/Arduino/tree/master/libraries/Servo

про таймеры что-то есть здесь
https://github.com/arduino/Arduino/blob/master/libraries/Servo/src/avr/ServoTimers.h

1i7, 2016-12-25:
работа с таймерами в Ардуино

Использование прерываний в Arduino (продолжение)
http://arduino.ru/Tutorial/Arduino_Interrupts_part2

Arduino Timer and Interrupt Tutorial
https://oscarliang.com/arduino-timer-and-interrupt-tutorial/

1i7, 2016-12-25:
Черновой порт есть:
https://github.com/1i7/stepper_h/commit/badcfa78c13e890436f4cc2eedb3f3040651217b

Код компилируется, таймер тикает, мотор, вроде как, тоже шагает. Нужно только разобраться, какая там реальная частота и как ее можно регулировать.

1i7, 2017-01-02:
Так, вот эти тикеты не про таймеры, но прямое отношение к порту:

Разобраться с размерами целочисленных типов на разных архитектурах для step_delay
https://github.com/1i7/stepper_h/issues/13

Разобраться с размерами целочисленных типов на разных архитектурах для step_delay
https://github.com/1i7/stepper_h/issues/14

Тип данных для prepare_steps:step_count int -> long
https://github.com/1i7/stepper_h/issues/15

1i7, 2017-08-20:
Ладно, подробнее,

код для запуска таймера из Servo.cpp выглядит примерно так:
https://github.com/arduino/Arduino/blob/ide-1.5.x/libraries/Servo/src/avr/Servo.cpp

~~~cpp
void initTimerISR(int timer, int prescalar, int period)
{
#if defined (_useTimer1)
  if(timer == _timer1) {
    TCCR1A = 0;             // normal counting mode
    TCCR1B = _BV(CS11);     // set prescaler of 8
    TCNT1 = 0;              // clear the timer count
#if defined(__AVR_ATmega8__)|| defined(__AVR_ATmega128__)
    TIFR |= _BV(OCF1A);      // clear any pending interrupts;
    TIMSK |=  _BV(OCIE1A) ;  // enable the output compare interrupt
#else
    // here if not ATmega8 or ATmega128
    TIFR1 |= _BV(OCF1A);     // clear any pending interrupts;
    TIMSK1 |=  _BV(OCIE1A) ; // enable the output compare interrupt
#endif
#if defined(WIRING)
    timerAttach(TIMER1OUTCOMPAREA_INT, Timer1Service);
#endif
  }
#endif

#if defined (_useTimer3)
  if(timer == _timer3) {
    TCCR3A = 0;             // normal counting mode
    TCCR3B = _BV(CS31);     // set prescaler of 8
    TCNT3 = 0;              // clear the timer count
#if defined(__AVR_ATmega128__)
    TIFR |= _BV(OCF3A);     // clear any pending interrupts;
	ETIMSK |= _BV(OCIE3A);  // enable the output compare interrupt
#else
    TIFR3 = _BV(OCF3A);     // clear any pending interrupts;
    TIMSK3 =  _BV(OCIE3A) ; // enable the output compare interrupt
#endif
#if defined(WIRING)
    timerAttach(TIMER3OUTCOMPAREA_INT, Timer3Service);  // for Wiring platform only
#endif
  }
#endif

#if defined (_useTimer4)
  if(timer == _timer4) {
    TCCR4A = 0;             // normal counting mode
    TCCR4B = _BV(CS41);     // set prescaler of 8
    TCNT4 = 0;              // clear the timer count
    TIFR4 = _BV(OCF4A);     // clear any pending interrupts;
    TIMSK4 =  _BV(OCIE4A) ; // enable the output compare interrupt
  }
#endif

#if defined (_useTimer5)
  if(timer == _timer5) {
    TCCR5A = 0;             // normal counting mode
    TCCR5B = _BV(CS51);     // set prescaler of 8
    TCNT5 = 0;              // clear the timer count
    TIFR5 = _BV(OCF5A);     // clear any pending interrupts;
    TIMSK5 =  _BV(OCIE5A) ; // enable the output compare interrupt
  }
#endif
}
~~~

Здесь почти одинаковые блоки для разных таймеров, по вызову кода получается так, что стартует Timer1

~~~cpp
[... 89 lines cut ...]
~~~

стало
~~~cpp
  TCCR1B = 0;
  TCCR1B |= (1 << WGM12);   // CTC mode
  TCCR1B |= (1 << CS12);    // 256 prescaler 
~~~

здесь важно, что выставлен режим CTC (Clear timer on compare match) - без него ничего работать не будет.

Плюс, т.к. в одном регистре настроечных битов несколько, значение не перезаписывается присвоением '=', а накладывается побитовым или '|='

После выставления этих настроек, значение в регистре OCR1A начинает влиять на частоту прерываний вместе с прескайлером, период можно полноценно регулировать.

Макрос _BV(CS11) идентичен бобитовому сдвигу (1 << CS11)

http://www.atmel.com/webdoc/AVRLibcReferenceManual/group__avr__sfr_1ga11643f271076024c395a93800b3d9546.html

~~~cpp
#define _BV(bit) \
    (1 << (bit))
~~~

этот синтаксис роли не играет

1i7, 2017-08-20:
Кстати, в Servo.cpp в initISR для таймера выставляется только прескалер (=8), а дополнительный счетчик выставляется динамически на нужное количество тиков перед каждым новым прерыванием внутри текущего прерывания handle_interrupts

https://github.com/arduino/Arduino/blob/ide-1.5.x/libraries/Servo/src/avr/Servo.cpp
~~~cpp
*OCRnA = *TCNTn + SERVO(timer,Channel[timer]).ticks;
~~~

1i7, 2017-08-21:
Запулил пример работы с таймером Ардуино для начала сюда
https://github.com/1i7/snippets/tree/master/arduino-timer/arduino_timer

главные моменты

включение таймера initTimerISR (внутри)
~~~cpp
void initTimerISR(int timer, int prescaler, int period)
...
#if defined (_useTimer1)
  if(timer == _timer1) {
    // CS12 CS11 CS10
    // 1/1:    001
    // 1/8:    010
    // 1/64:   011
    // 1/256:  100
    // 1/1024: 101
    unsigned char prescalerBits = 0;
    switch(prescaler) {
        case TIMER_PRESCALER_1_1:
            prescalerBits = _BV(CS10);
            break;
        case TIMER_PRESCALER_1_8:
            prescalerBits = _BV(CS11);
            break;
        case TIMER_PRESCALER_1_64:
            prescalerBits = _BV(CS11) | _BV(CS10);
            break;
        case TIMER_PRESCALER_1_256:
            prescalerBits = _BV(CS12);
            break;
        case TIMER_PRESCALER_1_1024:
            prescalerBits = _BV(CS12) | _BV(CS10);
            break;
        default:
            // let it be no prescaler by default:
            prescalerBits = _BV(CS10);
    }
    
    TCCR1A = 0;             // normal counting mode
    TCCR1B = 0;             // timer mode:
    TCCR1B |= _BV(WGM12);   // CTC mode
    TCCR1B |= prescalerBits;// set prescaler
    OCR1A = period;         // compare match register
    TCNT1 = 0;              // clear the timer count
#if defined(__AVR_ATmega8__)|| defined(__AVR_ATmega128__)
    TIFR |= _BV(OCF1A);      // clear any pending interrupts;
    TIMSK |=  _BV(OCIE1A) ;  // enable the output compare interrupt
#else
    // here if not ATmega8 or ATmega128
    TIFR1 |= _BV(OCF1A);     // clear any pending interrupts;
    TIMSK1 |=  _BV(OCIE1A) ; // enable the output compare interrupt
#endif
#if defined(WIRING)
    timerAttach(TIMER1OUTCOMPAREA_INT, Timer1Service);
#endif
  }
#endif
~~~

запуск таймера (снаружи)

~~~cpp
    // http://www.robotshop.com/letsmakerobots/arduino-101-timers-and-interrupts
    // 1. CPU frequency 16Mhz for Arduino
    // 2. maximum timer counter value (256 for 8bit, 65536 for 16bit timer)
    // 3. Divide CPU frequency through the choosen prescaler (16000000 / 256 = 62500)
    // 4. Divide result through the desired frequency (62500 / 2Hz = 31250)
    // 5. Verify the result against the maximum timer counter value (31250 < 65536 success) if fail, choose bigger prescaler.
    
    // Настроим и запустим таймер с периодом 20миллисекунд (50 срабатываний в секунду):
    // prescaler=1:64, period=0x61A8:
    // 16000000/8/50=40000 (50Hz - срабатывает 50 раз в секунду, т.е. каждые 20мс)
    // Обработчик прерывания от таймера - функция handle_interrupts 
    // (с заданными настройками будет вызываться каждые 20мс).
    initTimerISR(TIMER1, TIMER_PRESCALER_1_8, 40000);
~~~

1i7, 2017-08-23:
фубля, разобрался и оттестировал со всеми частотами

все улетело в отдельную библиотеку, работает с  Arduino AVR и ChipKIT PIC32 одновременно
https://github.com/1i7/arduino-timer-api

конкретно по работе с таймером и подбор параметров для разных частот на Arduino AVR
https://github.com/1i7/arduino-timer-api/blob/master/timer-api/timer_setup_avr.c

для ChipKIT PIC32
https://github.com/1i7/arduino-timer-api/blob/master/timer-api/timer_setup_pic32.c

для архива, здесь проект чисто для Arduino AVR:
https://github.com/1i7/snippets/tree/master/arduino-timer/arduino_timer
аналогичный проект для ChipKIT PIC32:
https://github.com/1i7/snippets/tree/master/chipkit-timer

Теперь нужно перенести код сюда в stepper_h или вообще перевести stepper_h на эту библиотеку.

sadr0b0t, 2017-08-25:
Провел замеры максимальной частоты на платах
https://github.com/sadr0b0t/arduino-timer-api/issues/4

получил
- PIC32MX 80MHz (ChipKIT Uno32): 200КГц (период 5мкс)
- AVR 16MHz (Arduino Leonardo): 20КГц (период 50мкс)

Для AVR 50КГц и 100КГц тоже в некотором роде рабочие, но начинают давать заметную погрешность.

Здесь же - добавил в примеры управление шаговым моторов в фоне таймеров (простейший вариант)
https://github.com/sadr0b0t/arduino-timer-api/blob/master/timer-api/examples/timer-api-stepper/timer-api-stepper.ino

sadr0b0t, 2017-08-29:
Порт есть https://github.com/sadr0b0t/stepper_h/commit/906d1e5974f10c8c5250c782594ba47c4fa8961b

Таймер тикает, шаги отсчитывает.
- Код управления таймером, настройки частот, вызова прерываний скопировал из https://github.com/1i7/arduino-timer-api (задействовать саму библиотеку без копирования кода по ряду нюансов API пока не получается и не факт, что это вообще нужно; при этом, из-за переноса определений кое-каких констант с без переименования, использовать библитеку timer-api совместно со stepper_h сейчас скорее всего не получится, но это при случае будет не очень сложно исправить)
- В stepper_timer.cpp (и дубль - в stepper_lib_config.h) попало немного платформенно-зависимого кода - в самом начале при установке настроек частоты таймера по умолчанию, подругому пока сделать не получается.

Замерил производительность для разного количества управляемых одновременно моторов: https://github.com/sadr0b0t/stepper_h/issues/25

Решил, что значения по умолчанию будут частоты, достаточные для вращения одновременно 3х моторов (добавил в ридми на глагне):
- PIC32MX/ChipKIT:  50КГц (период=20мкс)
- AVR/Arduino: 5КГц (период=200мкс) 

Помимо настроек таймера пришлось хорошо перелопатить типы данных, в основном:
- перейти с int на long (unsigned int -> unsigned long)
- расставить unsigned везде, где переменные или поля не подразумевают использование отрицательных значений

На 32-битном int на PIC32 во многих местах на эти нюансы можно было закрывать глаза, но на AVR int 16 бит и его максимального значения перестало хватать во многих даже не сильно граничных случаях.

Плюс, размер int разный на разных платформах, что порождает гемор и неразбириху, а long везде одинаковый и как минимум 32 бита (64-битные платформы пока не рассматриваем).

Отсюда правило (дубль отсюда https://github.com/sadr0b0t/stepper_h/issues/22 )
- 32 бит (тип данных long/unsigned long, 32 бит на AVR и PIC32) - везде, где используются единицы измерения из реального мира (длина или время) или значения, на них влияющие (количество шагов в цикле),
- 64 бит (тип данных long long, 64 бит на AVR и PIC32) - там, где не хватает long
- 16/32 бит (тип данных int, 16 бит на AVR, 32 бит на PIC32) - для вспомогательной целочисленной мелочевки (номера пинов, индексы массивов и т.п.)

Тесты проходят, порт на первый взгляд рабочий. Хотя, пример draw_triangle после нескольких циклов рисования может начать выдавать странные значения, причем почти всегда только на координате x, нужно разобраться, почему так происходит, но уже в отдельном тикете.

sadr0b0t, 2017-08-29:
коменты в ридми вдогонку https://github.com/sadr0b0t/stepper_h/commit/fea567530f83fecec2db083afd4f416ab6b8d04e

closed 2017-08-29 by sadr0b0t

sadr0b0t, 2017-08-29:
Да, еще любопытный нюанс при переходе к unsigned long - в stepper_timer.cpp пришлось добавить вот такое явное приведение типов:

~~~cpp
if( _cstatuses[i].calibrate_mode == NONE &&
                        (_cstatuses[i].dir > 0 ?
                            _smotors[i]->max_end_strategy != INF &&
                                _smotors[i]->current_pos + (long long)_smotors[i]->distance_per_step > _smotors[i]->max_pos :
                            _smotors[i]->min_end_strategy != INF &&
                                _smotors[i]->current_pos - (long long)_smotors[i]->distance_per_step < _smotors[i]->min_pos) ) {
~~~

и еще похожая ситуация недалеко в другом месте

здесь _smotors[i]->current_pos - тип long long (64 бит со знаком), _smotors[i]->distance_per_step - unsigned long (32 бит без знака), _smotors[i]->min_pos - long long (64 бит со знаком).

Изначально (до того, как distance_per_step стал unsigned) никакого приведения типов не было и всё работало.

После добавления unsigned проверка поломалась - к счастью, из-за нее проваливался тест (выход за границу влево меньше 0), который опять начал проходить после добавления явного приведения типа (или при удалении unsigned от distance_per_step). Без провалившегося теста можно было бы еще очень долго ловить багу.

sadr0b0t, 2017-09-02:
Последний штрих - внезапно выяснилось, что timer_stop_ISR не работал, теперь работает.
https://github.com/sadr0b0t/stepper_h/issues/26

Теперь порт полностью рабочий